tessel/t2-firmware #244: Port socket errors when writing and reading pins at roughly the same time

Lange, 2018-10-22:
Hello 👋 

## Summary
When a program writes and reads to pins in close temporal proximity, there is a chance for errors such as the following to arise:
```
/usr/lib/node/tessel-export.js:429
              throw new Error('Received unexpected data packet');
              ^

Error: Received unexpected data packet
    at Socket.Port.sock.on (/usr/lib/node/tessel-export.js:429:21)
    at emitNone (events.js:106:13)
    at Socket.emit (events.js:208:7)
    at emitReadable_ (_stream_readable.js:513:10)
    at emitReadable (_stream_readable.js:507:7)
    at addChunk (_stream_readable.js:274:7)
    at readableAddChunk (_stream_readable.js:250:11)
    at Socket.Readable.push (_stream_readable.js:208:10)
    at Pipe.onread (net.js:597:20)
```

## Context
I encountered this error while working on https://github.com/TipoftheHats/tally-lights. I joined the community Slack and spoke with @HipsterBrown, who helped me debug the issue. We discovered a somewhat minimal repro case, which I was asked to post here in a new issue.

## Repro steps
Run the following code on a T2 with `t2 run <filename>.js`. Let it run until it errors, or outright stalls. It might happen immediately, it might happen after a minute, or it might not happen at all. In my experience, there's usually an error or stall within 2-3 minutes.

```js
'use strict';

const tessel = require('tessel');
const NUM_PINS_PER_PORT = 8;

function readAllPins() {
	console.log('Reading all pins...');
	for (const portName in tessel.port) {
		const port = tessel.port[portName];
		for (let i = 0; i < NUM_PINS_PER_PORT; i++) {}
		port.pin.forEach((pin, pinIndex) => {
			pin.read((error, value) => {
				if (error) {
					console.error('Failed to read pin:', error);
				} else {
					console.log(`Port ${portName}, pin #${pinIndex}: ${value}`)
				}
			});
		});
	}
}

let lastValue = 0;
function writeToOnePin() {
	const newValue = lastValue === 0 ? 1 : 0;
	tessel.port.A.pin[0].write(newValue, (error => {
		if (error) {
			console.error('Failed to write pin:', error);
		} else {
			console.log(`Successfully wrote ${newValue} to pin`);
		}
	}));
	lastValue = newValue;
}

setInterval(() => {
	readAllPins()
}, 3000);

setInterval(() => {
	writeToOnePin();
}, 1000);
```

This repro could likely be reduced down even further, but this was the best I could do with short time.

HipsterBrown, 2018-10-27:
@Lange So I got some time to run this example and was not able to recreate the error. Can you include the version info for your Tessel? Run `t2 version` with your Tessel connected to your computer.

Lange, 2018-11-03:
Hi @HipsterBrown, sorry for the delay. Here is the output of `t2 version` with my Tessel connected:

```
INFO Looking for your Tessel...
INFO Connected to Tessel-02A396939177.
INFO Tessel Environment Versions:
INFO t2-cli: 0.1.20
INFO t2-firmware: 0.2.0
INFO Node.js: 8.11.3
```